daisyaudio/DaisyExamples #31: DaisyExamples daisysp.h missing in some cases, cannot find -ldaisysp

ggsubs, 2020-06-02:
When building on Mac, the DaisyExamples repo does not build the examples. DaisySP and libs are building fine. Attached the output of rebuild_all.sh.

Two issues:
1. ld: cannot find -ldaisysp
2. MusicBox.cpp:2:10: fatal error: daisysp.h: No such file or directory

```
./rebuild_all.sh 
/Volumes/HP950/dev/DaisyExamples
rebuilding everything. . . 
only errors, and warnings will output. . . 
-------------------
rebuilding libdaisy
src/per_qspi.c:658:16: warning: 'get_status_register' defined but not used [-Wunused-function]
  658 | static uint8_t get_status_register(QSPI_HandleTypeDef *hqspi)
      |                ^~~~~~~~~~~~~~~~~~~
src/per_qspi.c:627:17: warning: 'exit_quad_mode' defined but not used [-Wunused-function]
  627 | static uint32_t exit_quad_mode(QSPI_HandleTypeDef *hqspi)
      |                 ^~~~~~~~~~~~~~
src/per_qspi.c:598:17: warning: 'enter_quad_mode' defined but not used [-Wunused-function]
  598 | static uint32_t enter_quad_mode(QSPI_HandleTypeDef *hqspi)
      |                 ^~~~~~~~~~~~~~~
src/per_sai.c: In function 'HAL_SAI_MspInit':
src/per_sai.c:312:34: warning: unused variable 'pSyncConfig' [-Wunused-variable]
  312 |     HAL_DMA_MuxSyncConfigTypeDef pSyncConfig;
      |                                  ^~~~~~~~~~~
src/hid_oled_display.cpp:89:21: warning: 'cfg_dc' defined but not used [-Wunused-variable]
   89 | static dsy_gpio_pin cfg_dc    = {DSY_GPIOB, 4};
      |                     ^~~~~~
src/hid_oled_display.cpp:88:21: warning: 'cfg_reset' defined but not used [-Wunused-variable]
   88 | static dsy_gpio_pin cfg_reset = {DSY_GPIOB, 15};
      |                     ^~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c: In function 'USB_WritePacket':
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c:934:7: warning: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Wattributes]
  934 |       USBx_DFIFO((uint32_t)ch_ep_num) = *((__packed uint32_t *)pSrc);
      |       ^~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c: In function 'USB_ReadPacket':
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_ll_usb.c:963:5: warning: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Wattributes]
  963 |     *(__packed uint32_t *)pDest = USBx_DFIFO(0U);
      |     ^
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c: In function 'HAL_SPI_Transmit':
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:885:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  885 |           *((__IO uint16_t *)&hspi->Instance->TXDR) = *((uint16_t *)hspi->pTxBuffPtr);
      |            ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:924:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  924 |           *((__IO uint16_t *)&hspi->Instance->TXDR) = *((uint16_t *)hspi->pTxBuffPtr);
      |            ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c: In function 'HAL_SPI_Receive':
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:1099:47: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 1099 |           *((uint16_t *)hspi->pRxBuffPtr) = *((__IO uint16_t *)&hspi->Instance->RXDR);
      |                                              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:1139:47: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 1139 |           *((uint16_t *)hspi->pRxBuffPtr) = *((__IO uint16_t *)&hspi->Instance->RXDR);
      |                                              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c: In function 'HAL_SPI_TransmitReceive':
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:1330:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 1330 |           *((__IO uint16_t *)&hspi->Instance->TXDR) = *((uint16_t *)hspi->pTxBuffPtr);
      |            ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:1349:47: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 1349 |           *((uint16_t *)hspi->pRxBuffPtr) = *((__IO uint16_t *)&hspi->Instance->RXDR);
      |                                              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:1388:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 1388 |           *((__IO uint16_t *)&hspi->Instance->TXDR) = *((uint16_t *)hspi->pTxBuffPtr);
      |            ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:1414:47: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 1414 |           *((uint16_t *)hspi->pRxBuffPtr) = *((__IO uint16_t *)&hspi->Instance->RXDR);
      |                                              ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c: In function 'HAL_SPI_IRQHandler':
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_spi.c:2786:49: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 2786 |             *((uint16_t *)hspi->pRxBuffPtr) = *((__IO uint16_t *)&hspi->Instance->RXDR);
[... 164 lines cut ...]
done
rebuilding petal/PedalTemplate/
/Volumes/HP950/dev/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot find -ldaisysp
collect2: error: ld returned 1 exit status
make: *** [../../libdaisy/core/Makefile:186: build/ex_patch_pluck_echo.elf] Error 1
done
rebuilding petal/template/
/Volumes/HP950/dev/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot find -ldaisysp
collect2: error: ld returned 1 exit status
make: *** [../../libdaisy/core/Makefile:186: build/ex_petal_template.elf] Error 1
done
rebuilding field/KeyboardTest/
KeyboardTest.cpp:2:10: fatal error: daisysp.h: No such file or directory
    2 | #include "daisysp.h"
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [../../libdaisy/core/Makefile:180: build/KeyboardTest.o] Error 1
done
rebuilding field/template/
/Volumes/HP950/dev/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot find -ldaisysp
collect2: error: ld returned 1 exit status
make: *** [../../libdaisy/core/Makefile:186: build/ex_field_template.elf] Error 1
done
finished
```

stephenhensley, 2020-06-02:
@ggsubs thanks for the details! looks like this is a case issue. 

We'll push a fix ASAP, but for now you can band-aid this by changing the `daisysp` in `rebuild_libs.sh` and `rebuild_examples.sh` to `DaisySP`. 

This may need to be done in the Makefile of the examples you're trying to compile.

Windows is case-insensitive so it wouldn't have gotten caught there, and previously the repo was named `daisysp` instead, which would have also prevented this from being caught on older clones of the repo.

stephenhensley, 2020-06-02:
Tested on OS X Catalina

closed 2020-06-02 by stephenhensley