daisyaudio/DaisyExamples #33: Daisy Seed SimpleADC

monkeyharhar, 2020-06-02:
I am trying to get the SimpleADC example to work properly with the Daisy seed.

I have a 10k potentiometer with the wiper connected to pin D15 (ADC INP10) with the corresponding legs connected to 3v3_A and ground. I am using a multi-meter and can guarantee that I get a range of 0v to 3v3_A on the header pin. 

The LED just keeps blinking away at a constant rate seeming not to care about any value from the ADC. I was originally trying to hook up the pot to control freq of an oscillator of another example and could not get that to work so I figured I would try the simple example first.

I have tried rebuilding the example using other ADC pins (D16 and D19) with no success. 

Sorry if this is not the correct area to be asking for this sort of help.

stephenhensley, 2020-06-03:
@monkeyharhar This is the perfect place to ask these sorts of things!

Can you share a little about your toolchain/OS so that I can try to replicate what you're experiencing.

I just verified that this example is working on my Windows machine, and a fresh clone of the DaisyExample repo on Mac OS X.

(the LED will blink at a moderate rate with the knob all the way up, and really fast with the knob all the way down.)

monkeyharhar, 2020-06-03:
@stephenhensley Of course! I am running Windows 10, using the toolchain described in the Getting Started wiki (ARM toolchain, dfu-util). I've built the example using the make file. 

Here is a picture of my breadboard circuit. I've got the +/- rails hooked up to 3v3_A and GND, the 10k pot is hooked up to 3v3_A, D15, and GND, in that order. The output of the pot tested off and on pin is the correct range (0v - ~4.1v). 

[image]

stephenhensley, 2020-06-03:
Perfect. Thanks for the info, and the photo actually is the biggest help.

The ADCs use AGND as their ground reference, and for proper grounding you're going to want to join AGND and DGND outside of the Daisy.

You can do this by adding another jumper wire from pin 20 to your shared GND column on the breadboard.

We'll be publishing some hardware recommendations very soon that will cover some of the stuff like this, with wiring diagrams. 

Hope that helps! If that fixes, let me know and I'll close the issue.

monkeyharhar, 2020-06-03:
@stephenhensley Thank you very much! After jumping pin 20 to my GND rail, it now works as stated!

I am looking forward to the wiring diagrams. 

I would recommend changing the Wiring Guide in the SimpleAdc.cpp as it states:
// Connect other legs of pot to GND and 3v3_A respectively.

closed 2020-06-03 by monkeyharhar