daisyaudio/DaisyExamples #34: Daisy Pod/Patch/etc schematics

monkeyharhar, 2020-06-07:
I've currently been combing through the other examples for the different configurations and there are some examples that I'd like to use with the Daisy Seed.

I've currently been going to the header and class file for the Patch configuration to figure out what potentiometers are wired to which pins to try and create the configuration on a bread board.

Will the schematics or pin-outs be available for each configuration so we can attempt to wire one up ourselves?

stephenhensley, 2020-06-09:
@monkeyharhar 

We have added a Hardware repo that will be home to schematics, etc. for all of the breakout boards and more.

For now, only the Daisy Seed, and the Daisy Patch schematics are up, but we'll be adding the others soon.

https://github.com/electro-smith/Hardware

stephenhensley, 2020-06-11:
closing since the repo is up. Other schematics coming soon.

closed 2020-06-11 by stephenhensley