daisyaudio/DaisyExamples #83: Patch - MIDI Example doesn't make sound

stephenhensley, 2020-08-06:
Its possible whatever cutoff isn't getting set, or something else as well.

makingsoundmachines, 2020-08-10:
I've had a go at it and this works for me. I took the display code out, you might want to add it back in. Added in a Seed LED flash on Note on.

`#include "daisy_patch.h"
#include "daisysp.h"
#include <string>

using namespace daisy;
using namespace daisysp;

DaisyPatch  patch;
MidiHandler midi;
Oscillator  osc;
MoogLadder  filt;

bool led_state = false;

void AudioCallback(float **in, float **out, size_t size)
{
    
    for(size_t i = 0; i < size; i += 2)
    {
        
        for (size_t chn = 0; chn < 4; chn++)
        {
            float saw = osc.Process();
            float sig_out = filt.Process(saw);

            out[chn][i] = sig_out; //filt.Low()
        }
    }
}

// Typical Switch case for Message Type.
void HandleMidiMessage(MidiEvent m)
{
    switch(m.type)
    {
        case NoteOn:
        {
            NoteOnEvent p = m.AsNoteOn();
            // This is to avoid Max/MSP Note outs for now..
            if(m.data[1] != 0)
            {
                p = m.AsNoteOn();
                osc.SetFreq(mtof(p.note));
                osc.SetAmp((p.velocity / 127.0f));
            }
            
            led_state = true;
            patch.seed.SetLed(led_state);
        }
        break;
        case NoteOff:
        {
            led_state = false;
            patch.seed.SetLed(led_state);
        }
        break;
        case ControlChange:
        {
            ControlChangeEvent p = m.AsControlChange();
            switch(p.control_number)
            {
                case 1:
                    // CC 1 for cutoff.
                    filt.SetFreq(mtof((float)p.value));
                    break;
                case 2:
                    // CC 2 for res.
                    filt.SetRes(((float)p.value / 127.0f));
                    break;
                default: break;
            }
            break;
        }
        default: break;
    }
}

// Main -- Init, and Midi Handling
int main(void)
{
    // Init
    float sample_rate;
    patch.Init();
    sample_rate = patch.AudioSampleRate();
    
    dsy_system_delay(250);
    midi.Init(MidiHandler::INPUT_MODE_UART1, MidiHandler::OUTPUT_MODE_NONE);

    // Synthesis
    osc.Init(sample_rate);
    osc.SetWaveform(Oscillator::WAVE_POLYBLEP_SAW);
    osc.SetFreq(220);
    osc.SetAmp(0.25);

    filt.Init(sample_rate);
    filt.SetFreq(5000);
    filt.SetRes(0.5);

    patch.seed.SetLed(led_state);
    
    // Start stuff.
    patch.StartAdc();
    patch.StartAudio(AudioCallback);
    midi.StartReceive();
    for(;;)
    {
        midi.Listen();
        // Handle MIDI Events
        while (midi.HasEvents())
        {
            HandleMidiMessage(midi.PopEvent());
        }
    }
}`

makingsoundmachines, 2020-08-10:
@stephenhensley the strange thing is it started working when I added in the LED code -

patch.seed.SetLed(led_state);

maybe there's an addional clue to what's up in the original example.

stephenhensley, 2020-08-14:
It was a missing initializer in the `chn` loop within the audio callback. 

Resolved in #85

closed 2020-08-14 by stephenhensley