daisyaudio/DaisyExamples #126: Make Create New Project Python Script

andrewikenberry, 2020-12-14:


andrewikenberry, 2020-12-14:
[image] [Create New Project Python Script](https://trello.com/c/PvKL9Zif/64-create-new-project-python-script)

stephenhensley, 2020-12-14:
Just weighing in before the PR shows up, but this is much needed especially as Debug support gets added to all the examples.

After doing a very brief amount of playing around with VCV rack a week or so ago, it was cool to see their python helper script, `helper.py`, be a singular script to assist with most utility actions. Might be worth doing something similar here to limit the number of scripts needed.

could be like:

```
#  Create MyNewProject for DaisyPatch
python helper.py create --board patch MyNewProject

# Duplicate MyOldProject and and rename everything to MyNewProject
python helper.py duplicate MyOldProject MyNewProject
```

the long-form  `--board` could be `-b` or just a second positional argument.

stephenhensley, 2020-12-23:
works as part of helper.py

closed 2020-12-23 by stephenhensley