daisyaudio/DaisyExamples #174: Hi,

paoluzzi, 2021-02-06:


paoluzzi, 2021-02-06:
Hi, is there a way to change the PWM frequency? I would like to move it from 1 kHz to higher values.

paoluzzi, 2021-02-06:
... I am using a Daisy Seed.

stephenhensley, 2021-02-09:
The samplerate optional parameter controls the frequency of the software PWM.

You can set it by creating a new Led class, and initializing it with the desired frequency. You do have to call the `Update` function at that frequency for it to work as expected.

I am going to be reworking this soon to provide a background timer or something a bit easier. And certain pins are capable of hardware PWM where the frequency can be set explicitly, but this is not present in libdaisy yet.

In the future, questions like this may receive a faster answer on either the forum or in the libdaisy github issues (since this isn't directly related to the examples).

closed 2021-02-09 by stephenhensley