daisyaudio/DaisyExamples #176: Daisy Seed delayline example - change feedback calculation?

StaffanMelin, 2021-02-13:
I studied the example to use it myself.

Wouldn't it make more sense to change the line

        feedback = (del_out * 0.75f) + osc_out;

to

        feedback = (del_out +  + osc_out) * 0.75;

so that the "original" sound (osc_out) is reduced together with the old sound (del_out) before written to the delayline. Else we get that one at the same strength a second time? (Hard to explain in words.)

At least this makes more sense sound-wise to me in my application when I use it as an insert effect. :)

stephenhensley, 2021-03-02:
As an insert effect,  you may want to leave the mix full-wet instead of attenuating the input along with the feedback.

Traditionally a delay effect with 0 feedback will produce a single delayed copy of the input at the interval set by the delay time.

The feedback is what would allow that to create the decaying-echo effect. In either case if you're only listening to the output of the delay line you would not hear the dry input signal at all.

My calling that variable "feedback" is a bit misleading as it is the feedback and the input summed together. A more legible version might look like:

```c++
del_out = del.Read();
feedback = del_out * fb_amt;
del.Write((dry_input * send_amt) + feedback);
```

In your use-case you may want to have a "send amount" instead of a dry/wet mix as illustrated above.

closed 2021-03-02 by stephenhensley

StaffanMelin, 2021-03-03:
Thank you for taking the time to explain! :)

stephenhensley, 2021-03-04:
No problem!