daisyaudio/DaisyExamples #215: Examples do not build

ri0t, 2021-09-09:
I get a lot of warnings when building libDaisy, but it succeeds, none the less.
When `rebuild_all.sh` starts building the examples, all fail. One shorter example would be SimpleOscillator:

```
21-09-09 3:32:16 with riot in ~/src/RESiSTY/DaisyExamples/pod/SimpleOscillator on  master 
➜ make               
arm-none-eabi-g++  -c -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32H750xx -DUSE_HAL_DRIVER -DHSE_VALUE=16000000 -DSTM32H750xx -I../../libdaisy -I../../libdaisy/src/ -I../../libdaisy/Drivers/CMSIS/Include/ -I../../libdaisy/Drivers/CMSIS/Device/ST/STM32H7xx/Include -I../../libdaisy/Drivers/STM32H7xx_HAL_Driver/Inc/ -I../../libdaisy/core/ -I../../DaisySP  -O2 -Wall -Wno-missing-attributes -fasm -fdata-sections -ffunction-sections -MMD -MP -MF"build/SimpleOscillator.d" -fno-exceptions -fasm -finline -finline-functions-called-once -fshort-enums -fno-move-loop-invariants -fno-unwind-tables  -std=gnu++14 -Wa,-a,-ad,-alms=build/SimpleOscillator.lst SimpleOscillator.cpp -o build/SimpleOscillator.o
SimpleOscillator.cpp:24:13: error: variable or field 'AudioCallback' declared void
   24 | static void AudioCallback(AudioHandle::InterleavingInputBuffer  in,
      |             ^~~~~~~~~~~~~
SimpleOscillator.cpp:24:40: error: 'InterleavingInputBuffer' is not a member of 'daisy::AudioHandle'
   24 | static void AudioCallback(AudioHandle::InterleavingInputBuffer  in,
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~
SimpleOscillator.cpp:25:40: error: 'InterleavingOutputBuffer' is not a member of 'daisy::AudioHandle'
   25 |                           AudioHandle::InterleavingOutputBuffer out,
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~
SimpleOscillator.cpp:26:65: error: expected primary-expression before 'size'
   26 |                           size_t                                size)
      |                                                                 ^~~~
SimpleOscillator.cpp: In function 'int main()':
SimpleOscillator.cpp:78:19: error: 'AudioCallback' was not declared in this scope
   78 |     hw.StartAudio(AudioCallback);
      |                   ^~~~~~~~~~~~~
SimpleOscillator.cpp: At global scope:
SimpleOscillator.cpp:20:14: warning: 'freq' defined but not used [-Wunused-variable]
   20 | static float freq;
      |              ^~~~
make: *** [../../libdaisy/core/Makefile:195: build/SimpleOscillator.o] Error 1
```

Any idea what is going on? I'm on Debian with build-essential==12.9 and  gcc-arm-none-eabi-10.3-2021.07-x86_64-linux.tar.bz2

ri0t, 2021-09-09:
Same with gcc-arm-none-eabi-9-2019-q4-major

ri0t, 2021-09-09:
Umm. Those errors come from a very old git-submodule'd libdaisy. When i upgrade to the latest, it suddenly doesn't find `daisysp.h`

ri0t, 2021-09-09:
Yeah, because that repo's submodule pin is also outdated.. :unamused:

ri0t, 2021-09-09:
Well. Everything except "patch_sm" built now. Maybe updating the submodules could close this issue :grimacing:

stephenhensley, 2021-09-09:
@ri0t apologies for the inconvenience! Something triggered the submodules to jump back several months.

I just pushed the most recent submodules references and everything should be building again.

closed 2021-09-09 by stephenhensley