daisyaudio/DaisyExamples #220: libDaisy submodule capitalization

CorvusPrudens, 2021-09-09:
The libDaisy capitalization is inconsistent across the tools in DaisyExamples. The submodule is set to always clone into `libdaisy`, and some of the `helper.py` commands create lower case libDaisy references. Since the repo name is libDaisy, it's probably best to get everything aligned with that capitalization.

This was particularly annoying for mac / linux users because the unix cpp configuration for vscode assumed a capital D.

stephenhensley, 2021-12-16:
this should be resolved, but I know people _may_ still run into an issue with previous clones (though at this point, even with wrong spelling it compiles on the mac OS machine I was testing on ... So we'll see I suppose).

I'll leave this open for a bit with this note:

**If you get a `can't find libDaisy` related error -- you can either manually rename the "libdaisy" folder to "libDaisy", or reclone the repo to resolve the issue.**

MrBlueXav, 2022-02-13:
Hello, I'm on a linux platform.
It's a problem when using helper.py for creating a new project.
The makefile won't work as it refers to "libdaisy" instead of libDaisy.
Easy to fix, though.

HeckHeckHeckHeck, 2024-09-25:
The macos filesystem is acutally case insensitive

CorvusPrudens, 2024-09-25:
@HeckHeckHeckHeck It would be more accurate to say that APFS _can_ be case-sensitive, but it is not by default. [See here for details](https://support.apple.com/lt-lt/guide/disk-utility/dsku19ed921c/mac).

HeckHeckHeckHeck, 2024-09-25:
True, you are right! thanks for pointing that out. i actually forgot that since, case-insensitive is the default.

stephenhensley, 2026-10-20:
Did a quick check after all recent changes, and don't see anything left that would cause issues w/ copying via helper.py, etc. due to the capitalization.

If I missed anything, feel free to re-open.

closed 2026-10-20 by stephenhensley