daisyaudio/DaisyExamples #242: incorrect or hard coded paths in the c_cpp_properites.json

GR8Instruments, 2021-12-06:
I loaded up the DaisyExamples at the folder level into VS.Code and got the following 7 errors:

[{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples",
	"severity": 4,
	"message": "Cannot find \"d:\\work\\programming\\dsp\\Eigen\\eigen-3.3.8\".",
	"startLineNumber": 17,
	"startColumn": 17,
	"endLineNumber": 17,
	"endColumn": 60
},{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples",
	"severity": 4,
	"message": "Cannot find \"c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples\\DaisySP\\examples\\fastconv\".",
	"startLineNumber": 20,
	"startColumn": 17,
	"endLineNumber": 20,
	"endColumn": 63
},{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples",
	"severity": 4,
	"message": "Cannot find \"D:\\WORK\\PROGRAMMING\\DSP\\_DAISY\\DAISYEXAMPLES\\DAISYSP\\EXAMPLES\\FASTCONV\".",
	"startLineNumber": 22,
	"startColumn": 17,
	"endLineNumber": 22,
	"endColumn": 89
},{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples",
	"severity": 4,
	"message": "Cannot find \"c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples\\DaisySP\\tests\\fastconv\".",
	"startLineNumber": 23,
	"startColumn": 17,
	"endLineNumber": 23,
	"endColumn": 60
},{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples",
	"severity": 4,
	"message": "Cannot find \"c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples\\DaisySP\\modules\".",
	"startLineNumber": 25,
	"startColumn": 17,
	"endLineNumber": 25,
	"endColumn": 53
},{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "_generated_diagnostic_collection_name_#1",
	"severity": 4,
	"message": "String does not match the pattern of \"^\\d{2}\\.\\d{1}\\.\\d{5}\\.\\d{1}$|^8\\.1$\".",
	"startLineNumber": 39,
	"startColumn": 34,
	"endLineNumber": 39,
	"endColumn": 36
},{
	"resource": "/c:/Users/direc/Documents/GitHub/Daisy/DaisyExamples/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\direc\\Documents\\GitHub\\Daisy\\DaisyExamples",
	"severity": 4,
	"message": "Cannot find \"C:\\\\Program Files (x86)\\\\GNU Arm Embedded Toolchain\\\\9 2020-q2-update\\\\bin\\\\arm-none-eabi-g++.exe\".",
	"startLineNumber": 40,
	"startColumn": 29,
	"endLineNumber": 40,
	"endColumn": 128
}]

stephenhensley, 2021-12-16:
Thanks for bringing this up!

The root folder doesn't really have a use in VS code at the moment (each example has it's own workspace for compiling/debugging), but this definitely shouldn't be cluttered with peoples personal drive paths. 

We'll get this sorted out soon.

stephenhensley, 2026-10-20:
absolute paths have been removed (aside from the `compilerPath` which despite not existing, doesn't seem to cause any issues on non-windows systems). 
The defines, includePaths, and compilerArgs have been updated, and should no longer cause inaccurate error/warning messages.

closed 2026-10-20 by stephenhensley