daisyaudio/DaisyExamples #256: rebuild_all.sh crashes on Linux

znmeb, 2021-12-31:
`./rebuild_all.sh` crashes after building the libraries:

```
❯ ./rebuild_all.sh
building libDaisy . . .
arm-none-eabi-ar: creating build/libdaisy.a
done.
building DaisySP . . .
done.
./rebuild_all.sh: line 4: ./ci/build_examples.py: Permission denied
finished
```

The problem is that the script `ci/build_examples.py` needs to be executable.

stephenhensley, 2022-01-03:
oh interesting -- I had thought I resolved this already (because CI had the same issue). Are you on the most recent commit, or are you possibly a few commits behind?

If you have the permissions you can do `chmod +x ci/build_examples.py`.

znmeb, 2022-01-04:
I forked the whole repo right after I filed this issue, so I'm pretty sure I have the latest commit.

stephenhensley, 2022-01-04:
@znmeb good to know. I may have been thinking of libDaisy (we added a similar script there just before the holidays.

The DaisyExamples CI actually invokes python to call the script: `python ./ci/build_examples.py` so it wouldn't have caught this issue.

I'll merge a fix here in a few.

znmeb, 2022-01-04:
You have to be careful with using `python` like that. The system Python may be Python 2 or Python 3 and the user may be running in a virtual environment that has silently changed `PATH`.

commit 72a7591a (2022-02-02): fixes #256 - made build examples script executable (#257)
closed 2022-02-02 by stephenhensley