daisyaudio/DaisyExamples #260: empty file at /seed/Knob/resources/Knob.fzz

jordanmichaelfrancis, 2022-01-26:
The Fritzing file for the seed's knob example is empty:

[image]

When opened in Fritzing, the file produces the following error message:

> Unable to open '/Users/jordanfrancis/code/repos/DaisyExamples/seed/Knob/resources/Knob.fzz': zip.open(): %d 

[image]

stephenhensley, 2022-01-31:
Oh no! 

Thanks for letting us know, we'll look into this. And try to get it fixed ASAP.

stephenhensley, 2022-02-01:
@jordanmichaelfrancis thanks again for pointing this out. Should be resolved!

PR #262 (2022-02-01): Fix seed/knob fritzing file
Fixes #260
I went through and checked the other fritzing files and they all seem to be OK.
closed 2022-02-01 by stephenhensley