daisyaudio/DaisyExamples #277: Daisy Seed SDMMC.cpp example do not compile

KoSv, 2022-10-07:
I get errors trying to compile the simple sd card example.
`DaisyExamples/libDaisy//Middlewares/Third_Party/FatFs/src/ff.c:2698: undefined  reference to ff_convert`
`DaisyExamples/libDaisy//Middlewares/Third_Party/FatFs/src/ff.c:1693: undefined reference to `ff_wtoupper'`

and so on...

.... annoying

musicdevghost, 2022-11-03:
I'm experiencing the same issues and getting same error: "FatFs/src/ff.c:2698: undefined reference to `ff_convert'"

just with the simple test:

`/** SD card next */
    SdmmcHandler::Config sd_config;
    SdmmcHandler         sdcard;
    sd_config.Defaults();
    sdcard.Init(sd_config);

    fsi.Init(FatFSInterface::Config::MEDIA_SD);

    /** Write/Read text file */
    const char *test_string = "Testing Daisy Patch SM";
    const char *test_fname  = "DaisyPatchSM-Test.txt";
    FRESULT     fres = FR_DENIED; /**< Unlikely to actually experience this */
    if(f_mount(&fsi.GetSDFileSystem(), "/", 0) == FR_OK)
    {
        /** Write Test */
        if(f_open(&file, test_fname, (FA_CREATE_ALWAYS | FA_WRITE)) == FR_OK)
        {
            UINT   bw  = 0;
            size_t len = strlen(test_string);
            fres       = f_write(&file, test_string, len, &bw);
        }
        f_close(&file);
        if(fres == FR_OK)
        {
            /** Read Test only if Write passed */
            if(f_open(&file, test_fname, (FA_OPEN_EXISTING | FA_READ)) == FR_OK)
            {
                UINT   br = 0;
                char   readbuff[32];
                size_t len = strlen(test_string);
                fres       = f_read(&file, readbuff, len, &br);
            }
            f_close(&file);
        }
    }`

KoSv, 2022-11-03:
> I'm experiencing the same issues and getting same error: "FatFs/src/ff.c:2698: undefined reference to `ff_convert'"
> 
> 
> 
> just with the simple test:
> 
> 
> 
> 
> 
> SdmmcHandler::Config sd_config;
> 
>     SdmmcHandler         sdcard;
> 
>     sd_config.Defaults();
> 
>     sdcard.Init(sd_config);
> 
> 
> 
>     fsi.Init(FatFSInterface::Config::MEDIA_SD);
> 
> 
> 
>     /** Write/Read text file */
> 
>     const char *test_string = "Testing Daisy Patch SM";
> 
>     const char *test_fname  = "DaisyPatchSM-Test.txt";
> 
>     FRESULT     fres = FR_DENIED; /**< Unlikely to actually experience this */
> 
>     if(f_mount(&fsi.GetSDFileSystem(), "/", 0) == FR_OK)
> 
>     {
> 
>         /** Write Test */
> 
>         if(f_open(&file, test_fname, (FA_CREATE_ALWAYS | FA_WRITE)) == FR_OK)
> 
>         {
> 
>             UINT   bw  = 0;
> 
>             size_t len = strlen(test_string);
> 
>             fres       = f_write(&file, test_string, len, &bw);
> 
>         }
> 
>         f_close(&file);
> 
>         if(fres == FR_OK)
> 
>         {
> 
>             /** Read Test only if Write passed */
> 
>             if(f_open(&file, test_fname, (FA_OPEN_EXISTING | FA_READ)) == FR_OK)
> 
>             {
> 
>                 UINT   br = 0;
> 
>                 char   readbuff[32];
> 
>                 size_t len = strlen(test_string);
> 
>                 fres       = f_read(&file, readbuff, len, &br);
> 
>             }
> 
>             f_close(&file);
> 
>         }
> 
>     }

Try setting the Fs fat flag in the makefile.

stephenhensley, 2022-11-03:
Hi @KoSv , @musicdevghost 

sorry to hear you ran into an issue compiling this example.

If you're still running into this issue, can you make sure that your DaisyExamples repo is up to date (using `git pull origin master` ), and that your libDaisy submodule is on either the latest release version, or the current master branch, and rebuilt.

The current version of that example's makefile should does already contain the `USE_FATFS=1` mentioned above.

I tested locally, and verified that the CI properly compiled this example.

Feel free to reopen if you're still experiencing issues.

closed 2022-11-03 by stephenhensley