daisyaudio/DaisyExamples #279: I have an updated STMCubeIde example, should I upload it

jandyman, 2022-11-01:
I just got back to looking at Daisy, and updated an old example I have of using the STM32CubeIde for development, which I still think is the best option. It's based on the pod/SimpleOscillator example.

Should I upload it? File a PR? And since it is based on the SimpleOscillator example, would it make sense to put the project file there?

stephenhensley, 2022-11-03:
Hi @jandyman 

What does the file structure of the project look like.
I think an up-to-date example for STM32CubeIDE would be a welcome sight in the community, but I'm not sure if it would fit in this repo.

Let me know, and we can discuss where it might best fit in the ecosystem!

jandyman, 2022-11-06:
You know something? You got me thinking about how to use the existing makefiles within Eclipse (i.e. STM32CubeIde). Usually the problem with makefiles is that you have to update them manually, but in this case this is done by the daisy team anyway. 

It turns it out it's so simple that all I really need to convey is how to do it, there is no need to create an example project, and it can be done for any of the examples. So the revised question would be "where should I put the instructions"? 

I've also got instructions for building a native Eclipse project, which is sometimes easier to maintain as one moves away from the existing examples. 

@stephenhensley

stephenhensley, 2022-11-28:
Hi @jandyman that's an excellent thought, and would be a welcome addition to our resources!

We're currently planning out a long term plan for how to maintain documentation across the Daisy platform (as there is currently fairly sparse documentation in a lot of different places).

For the time being, the most portable way to add some documentation would be to add it to the libDaisy guides.
I'm not sure what making a PR on the DaisyWiki's wiki would look like, but that's another option. (We had to disable public editing of the wiki because links were getting edited by bots)

These guides are currently located in `libDaisy/doc/md`. Working with them as markdown makes it easy to preview locally, and load into something that can be included in the [web documentation](https://electro-smith.github.io/libDaisy/) or moved to other mediums with relative ease.

You can add your instructions as a new MD file there, and include a link to it in the `reference_home.md` file and include those changes in a PR on libDaisy.

Alternatively, you could make a new thread on the forum and include save yourself the PR part 😄 

Either way, let me know if you have any questions!