daisyaudio/DaisyExamples #281: 404 when linking a file from resources

ohmtech-rdi, 2022-12-19:
Some links in the documentation are broken.

For example in the [Drum](https://github.com/electro-smith/DaisyExamples/tree/master/seed/DSP/Drum) sample code:

The breadboard illustration links to 
```
https://raw.githubusercontent.com/electro-smith/DaisyExamples/master/seed/Drum/resources/Drum_bb.png
```

But it seems that GitHub has deprecated those kind of links, so this will return a 404.

It should be probably replaced with:
```
https://github.com/electro-smith/DaisyExamples/blob/master/seed/DSP/Drum/resources/Drum_bb.png
```

(Originally reported by K. Shepherd on Daisy Slack)

stephenhensley, 2022-12-22:
this has been resolved.

Thanks for making the issue!

closed 2022-12-22 by stephenhensley

ohmtech-rdi, 2022-12-23:
- For reference, fixed in #282