daisyaudio/DaisyExamples #306: libDaisy is out of date

lerouxb, 2024-08-11:
I'd like to update the libDaisy submodule inside DaisyExamples so I can use some of the new features from the latest version  in my project. I've tried doing the naive thing where I just checked out the latest master branch inside the submodule, but I can't get anything to compile. I'm getting:

```
building libDaisy . . .
make: *** No rule to make target `build/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal.o', needed by `build/libdaisy.a'.  Stop.
make: *** Waiting for unfinished jobs....
Failed to compile libDaisy
Building: pod/SynthVoice
cc1: fatal error: stm32h7xx.h: No such file or directory
compilation terminated.
make: *** [build/startup_stm32h750xx.o] Error 1
finished
```

It would be nice if there's some documentation somewhere on how to go about this since unless DiasyExamples gets updated regularly it would be nice to be able to do this ourselves. Given that the script to create a project puts it inside DaisyExamples, this is basically the way projects would consume anything added libDaisy.

HeckHeckHeckHeck, 2024-09-25:
Hya,
this is not a bug, in DaisyExamples. 
Please make sure your clone contains all submodules recursively.
Either:
* clone using: `git clone --recurse-submodules https://github.com/electro-smith/DaisyExamples`
* in the repo folder, add them: `git submodule update --init --recursive

`

lerouxb, 2024-09-30:
The submodules in DaisyExamples are still pinned to older commits of those submodules. DaisyExamples' master branch hasn't been updated in 8 months.

HeckHeckHeckHeck, 2024-09-30:
yes, the DaisyExamples submodules are on quite old versions (4.x something iirc) but again iirc they all the examples do compile correctly.

The only root cause i know for your problem problem description (at least popular root cause) is not having the submodules clones correctly.

Did you try the command line solutions i proposed?

HeckHeckHeckHeck, 2024-09-30:
i am trying to reproduce as we speak:
```bash
[heck@YOOROOX::~/tmp]  $ git clone --recurse-submodules https://github.com/electro-smith/DaisyExamples
Cloning into 'DaisyExamples'...
remote: Enumerating objects: 14396, done.
remote: Counting objects: 100% (3167/3167), done.
remote: Compressing objects: 100% (468/468), done.
remote: Total 14396 (delta 2826), reused 2699 (delta 2699), pack-reused 11229 (from 1)
Receiving objects: 100% (14396/14396), 598.66 MiB | 5.00 MiB/s, done.
Resolving deltas: 100% (10380/10380), done.
Submodule 'DaisySP' (https://github.com/electro-smith/DaisySP) registered for path 'DaisySP'
Submodule 'libDaisy' (https://github.com/electro-smith/libDaisy) registered for path 'libDaisy'
Submodule 'stmlib' (https://github.com/pichenettes/stmlib) registered for path 'stmlib'
Cloning into '/Users/heck/tmp/DaisyExamples/DaisySP'...
remote: Enumerating objects: 10836, done.
remote: Counting objects: 100% (3922/3922), done.
remote: Compressing objects: 100% (469/469), done.
remote: Total 10836 (delta 3574), reused 3666 (delta 3453), pack-reused 6914 (from 1)
Receiving objects: 100% (10836/10836), 7.99 MiB | 4.18 MiB/s, done.
Resolving deltas: 100% (9052/9052), done.
Cloning into '/Users/heck/tmp/DaisyExamples/libDaisy'...
remote: Enumerating objects: 36216, done.
remote: Counting objects: 100% (8671/8671), done.
remote: Compressing objects: 100% (1297/1297), done.
Receiving objects:  46% (16826/36216), 35.06 MiB | 3.80 MiB/s
```

lerouxb, 2024-10-10:
Yes I did. Those reset it back to the old versions. I'm trying to change the submodules to latest master of those repose.

atoav, 2024-12-25:
A very easy test to check if this works is to:

## 1. clone the Daisy repo

```bash
# 1. Clone the tepo
git clone --recurse-submodules https://github.com/electro-smith/DaisyExamples

# 2. Enter libDaisy directory
cd DaisyExamples/libDaisy

# 3. Update libDaisy to the latest master branch and switch to it
git pull origin master
git switch master

# 4. Try to rebuild all
./rebuild_all.sh
```

This produces the following output (errors):
```
building libDaisy . . .
make: *** No rule to make target 'build/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal.o', needed by 'build/libdaisy.a'.  Stop.
make: *** Waiting for unfinished jobs....
Failed to compile libDaisy
/usr/bin/env: ‘python’: No such file or directory
finished
```

stephenhensley, 2026-10-20:
Okay, this is resolved. Latest commits bump libDaisy up to v9.0.0, and update DaisySP as well.

libDaisy will continue to run slightly ahead of this repo (and DaisyBlinkProject), but will be kept up to date much more often–typically whenever libDaisy has a new release.

If you have an existing clone, instructions are identical to above:

```sh
git pull origin master
git submodule update --init --recursive
```

You should be able to rebuild libraries fine after this update (I tested both on a fresh clone and on an older version of the repo).

You may also want to run `git submodule sync --recursive` as the .gitmodules file has been updated to use the recently renamed "daisyaudio" org name. However, the old URL will continue to redirect without issues for the foreseeable future.

closed 2026-10-20 by stephenhensley